prusa3d/PrusaSlicer #12624: improper slicing on thin objects

petaflot, 2024-04-20:
### Description of the bug

check the cones on this temperature tower : only the middle one is correct
[image]

I guess the rest is self-explicit

### Project file & How to reproduce

STL is a temperature found on thingiverse

slice settings are 2 shells everywhere, does not depend on infill ratio or pattern.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version: 2.7.1 Build: PrusaSlicer-2.7.1+Gentoo

### Operating system

Gentoo

### Printer model

self build

neophyl, 2024-04-21:
You are probably slicing with the arachne engine,  adjust your settings.  Settings are not one value slices all.  

Although as you have not included the requested project file it is impossible to confirm this. You haven't even included the stl or a link so how is anyone supposed to reproduce this ?  I suspect you used this one - https://www.thingiverse.com/thing:4642743/files but there are hundreds, many almost identical

Slicing it with a default MK3 profile in classic gets you 
[image]
As you can see it slices fine.

With Arachne on some slightly adjusted settings I get the following
[image]
Again a slice that is fine.  

Everything depends on your settings.  Thinking that duplicating an 'issue' is obvious , just isnt.

petaflot, 2024-04-21:
OK all good, that was it. I didn't know about arachne and what it could do. Problem solved :-)

closed 2024-04-21 by petaflot