prusa3d/PrusaSlicer #12631: Incorrect handling of Minimum Fan Speed

OhGee3D, 2024-04-22:
### Description of the bug

I set the slicer to off for the first 2 layers and then max speed at layer 8 with minimum fan speed of 20% and max at 100%.

The slicer increases the speed from 0 at layer 2 to 20% at layer 8 instead of 0 at layer 2 to 100% at layer 8. It is incorrectly using the minimum instead of the max fan speed.

### Project file & How to reproduce

Set the filament settings as follows: -

Fan Speed -  Min: 20%   Max: 100%

Disable fan for the first 2 layers

Full fan speed at layer 8.

You will see the message "Fan speed will be ramped from zero at layer 2 to 20% at layer 8."

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Manjaro

### Printer model

Self Built

OhGee3D, 2024-04-22:
[fan-speed-error-project.zip](https://github.com/prusa3d/PrusaSlicer/files/15061545/fan-speed-error-project.zip)
[image]

rtyr, 2024-04-23:
That is a correct/intended behavior. Full fan speed is not the same value as max fan speed. Also you have "Enable auto cooling" disabled, so Max fan speed is ignored in general. Anyway, this feature just ramps up the speed gradually to "final" fan speed ("full" speed), if you have for example min 20, max 80, the "full fan speed" can be anywhere between those values based on the layer time.

closed 2024-04-23 by rtyr

OhGee3D, 2024-04-23:
I don't believe your response is correct. The ramping up starts **BELOW** the minimum fan speed and ramps up until it eventually reaches the "minimum" fan speed. So it starts below the speed which you have specified as the minimum speed your cooling fan is capable of with PWM. To use your own description from the Min tooltip "the minimum PWM your fan needs to work". It does **NOT** start at the minimum speed and work up to another value. It currently will only ever reach a maximum speed of the minimum value and all ramping up values are all below that figure. I suggest you reopen the case and investigate.

rtyr, 2024-04-23:
> The ramping up starts BELOW the minimum fan speed and ramps up until it eventually reaches the "minimum" fan speed.

Yes and that is exactly the point.  Lets say you have range min 50 to max 100. The real fan speed than depends on object size/layer time. Let say the speed will be chosen to 50%. If you will set full fan speed at layer 3 and disable fan for first layer then it will be as follows:

1st layer: fan disabled
2nd layer: fan 25%
3rd layer: fan 50% (full "targeet" fan speed).

The whole point of this feature is prevent the nozzle from cooling down rapidly at the beginning of the print. So even materials like PLA with min 100% max 100% have slower fan speed ramping.

OhGee3D, 2024-04-23:
So perhaps the wording for the tool-tip for the Min setting for fan speed needs adjusting? It is non-sensical as it stands. The tool-tip is as follows: -

**"This setting represents the minimum PWM your fan needs to work."** 

So, according to the Tooltip, the fan does not work at all below this value. With this in mind, it would be absolutely pointless (and potentially harmful to the fan which would be drawing current but not moving...) for the slicer to use any value below this.

Currently the slicer in no way treats this as any real minimum and is happy to set the fan way below it (in fact in this case  sets all ramp values value below or equal to it which seems absolutely crazy).

You must be able to see my confusion here. 

Anyway, sorry if it seem like I am nit-picking. Since we clearly disagree, I have pulled the code and changed it so it works in a sensible fashion for me. My printer is custom and so I have full freedom of the firmware and slicer I use.

I appreciate you are probably very busy and have more significant issues to deal with. But I am a little disappointed by your response.

rtyr, 2024-04-23:
I understand that this setting may be confusing, that is why I tried to explain how it works and what is it's purpose.

The legacy tooltip for min fan speed is indeed not great, we should change it.