prusa3d/PrusaSlicer #12632: Disable sequential printing in vase mode on XL.

NelisW, 2024-04-22:
### Description of the bug

Tried to print vase mode, multiple copies of the same  part. 
I did not realise that the X-axis gantry will knock/squash parts already printed.
For bed slingers the vertical sequential clearage makes sense, but not for coreXY: the bed moves and with it, all the parts.
Prevent the user from making this naive mistake - disable sequential printing for XL, or at least warn the user.

### Project file & How to reproduce

[vase-mode-container-v2-short-XL.zip](https://github.com/prusa3d/PrusaSlicer/files/15063216/vase-mode-container-v2-short-XL.zip)
The attached 3mf file will print the small caps in front because they are low enough, but fail on the printing the tall containers at the back.  In two of three test cases the bed actually titled by about 5-10 degrees, so this is a serious condition to be warned/disabled.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows 11

### Printer model

XL 5TH

u89djt, 2024-04-22:
(fellow user indicating an issue with your changed settings) Hi there, it looks like you've changed the clearance parameters such that the slicer thinks there will be sufficient clearance:
[image]
Reset those to their default values and the slicer will not let you do it:
[image]
This suggests - and I would agree - that the hover text of the parameters is not doing its job.

NelisW, 2024-04-22:
Hi Dave,

Thanks for replying so quickly!

I made the clearance values larger to be "more safe". So changes should not
be the cause.

On second figure: I did not get this warning. But even so, the sideway
clearance should be enough.

In any event, despite all above, the issue is not really the extruder
clearance. Rather the issue is the objects placed anywhere where the x
gantry can hit them. There is a similar issue in the MK3 and MK4, where the
x-rods can knock or squash the part.

I think the safest route is to at least warn the user that vase mode and
sequential print should not be used together.

Thanks for your trouble!

Nelis

u89djt, 2024-04-22:
Yeah, have another look at the hover text on the clearance values. Those numbers assert the radius of the extruder assembly, so a bigger value is safer, but the height asserts the height of the X axis carrier off the plate. By setting that to 45mm, you've said there is 45mm of clearance under the carriage /available/. You were hoping to tell the slicer how much space you /need/, but that's not what it does.
[image]
A bigger extruder radius is safer in many ways, but a bigger height entered into that box is definitely less safe.

u89djt, 2024-04-22:
Sticking my neck out here - would this do what you need? Scanning the layers in the slicing result makes it look like things should avoid each other, but we should certainly seek a second opinion from somewhere. 
[staggered vase-mode-container-v2-short-XL.zip](https://github.com/prusa3d/PrusaSlicer/files/15067339/staggered.vase-mode-container-v2-short-XL.zip)

u89djt, 2024-04-22:
Running from the back of the plate to the front keeps the X axis carriage out of the way, I think? Would this work out well in real life? Don't trust my judgement here, I just edited this now.
[staggered back to front vase-mode-container-v2-short-XL.zip](https://github.com/prusa3d/PrusaSlicer/files/15067411/staggered.back.to.front.vase-mode-container-v2-short-XL.zip)
[image]
[image]

NelisW, 2024-04-23:
Hi Dave,

Thanks, understood. User error, as always with things Prusa!

I took this as clearance around the part to be printed.

Thanks for spending your time on this!
I will close the bug report.

N

closed 2024-04-23 by rtyr

NelisW, 2024-04-23:
Hi Dave

Saw these two mails with the zip just now.
I will experiment with the two files and come back to you.
I just have to get past this workday (starting work now in South Africa).

Regards

N

u89djt, 2024-04-23:
No worries! It sounds like we're both doing this for the love of the game. Frontier spirit and all that.
I'm pretty sure the zip file in the comment I deleted will be bad news because it places the carriage on the side where there are parts already. I still think in MK3S :) The back to front ordering seems right when you slide the layer progress from nothing to a full plate imagining the carriage above it, so maybe that will work.

NelisW, 2024-04-23:
Hi Dave,

I switched on the (nozzle?) travel lines, and then is appears to be safe. I
can't test now, my solar battery is low (cloudy cold front passing over).
As soon as I have full solar again I will print and let you know.

Thanks, yet again!

Nelis

u89djt, 2024-04-23:
What do you think the hover text for the height setting should say?