prusa3d/PrusaSlicer #12639: M600 Filament Change or Height Modifiier Extruder Change with MMU Print

seanfrisbey, 2024-04-23:
I'm printing a project that I constantly successfully print. It is an MMU print, utilizing 2 colors. Recently, I've wanted to add a manual color change at a certain height in the print, and I don't want to have to go break the model into additional pieces just to facilitate this.

What I'm looking for is the ability to either set an M600 color change (like we can normally do in the sliced view) to change colors (I don't mind that it wouldn't use the wipe tower), or the ability to use a height modifier so set the appropriate print heights to use a different extruder (3 in this case).

The problem I'm running into it that it seems that MMU prints cannot be combined with either of these solutions. Using the M600 color change, I'm presented with an error reading "Note! There is a color change for extruder that won't be used until the end of print job. This code won't be processed during G-CODE generation.". Visually in the sliced view, it appears that this is correct, and the color change does not occur. Using the height modifier as an alternative yields no errors, but still doesn't reflect any difference in the sliced view.

It would be wonderful if either or both of these methods were possible when printing with an MMU setup.

Granted, maybe this is a problem only with the faux-MMU setup I have until I receive and upgrade my MMU2S -> MMU3, and this might work once I'm using a REAL MMU? But still, it would be nice for this to work in this situation as well.

I've attached an example 3mf file for this.

[ColorChangeTest.zip](https://github.com/prusa3d/PrusaSlicer/files/15082218/ColorChangeTest.zip)

drakon64, 2025-05-13:
To add to this, setting the extruder to be used within a height modifier to force a colour change is also ignored by PrusaSlicer

neophyl, 2025-05-13:
For drakon64 who says that the colour isn't applied for a height range modifier, well I just tried a cube added to Seans project with a height range modifier set to a different colour and it slices it with that different extruder.  

[image]

Perhaps you could attach your zipped up problem project where its not doing it.  I'm sure someone would take a look at it.

Checked with 2.9.2 on a windows 10 install.

drakon64, 2025-05-13:
That's interesting, I wonder if it's because I was enabling multi-material on a standard CORE One profile rather than using an MMU profile, or because there was a text modifier with a colour change inside of the height modifier

closed 2026-09-29 by prusaslicer-bot[bot]