prusa3d/PrusaSlicer #12641: Unauthorized while MK4 PrusaLink setup

ggodlewski, 2024-04-24:
### Description of the bug

After setup of physical printer I'm getting "Could not connect to PrusaLink: HTTP 401: 401: Unauthorized" when I press test button.

### Project file & How to reproduce

Printer -> edit physical printer

[image]

Press test.

I'm able to login using this password with the browser.
Uploading and printing through browser works.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+linux-x64-GTK3

### Operating system

Ubuntu 22.04

### Printer model

MK4

Jan-Soustruznik, 2024-04-25:
Hello, @ggodlewski , Cound you check if the slicer has grant right for work with network? Did you try to run PrusaSlicer with sudo?

ggodlewski, 2024-04-25:
I did factory reset and it worked with new password. Strange, I'm certain I copied password from password manager. Closing.