prusa3d/PrusaSlicer #12642: Bridge infill and first solid layer have the same alignment (Version 2.7.4+win64)

juckendesAuge, 2024-04-24:
### Description of the bug

“Solid Infill” goes in the same direction as “Bridge Infill”. Wouldn't diagonal be more suitable so that the first “solid infill” layer that belongs to the 4 “top layers” already looks better and is therefore probably also more stable?

[image]
[image]
[diagonal Skadis_for_prusa_MK3S.zip](https://github.com/prusa3d/PrusaSlicer/files/15095142/diagonal.Skadis_for_prusa_MK3S.zip)

### Project file & How to reproduce

Unzip the .zip.
Open the 3mf File.
slice it and check Bridge infill at Layer 3.80
and Solid infill at Layer 4.20

Prusa Slicer Version 2.7.4+win64

For information only:
I have added “M310 S0” to the Start G-Code under Printer Settings at the very end, as I always get a thermal anomaly warning with 0.6 and 0.8mm nozzles. With 0.4mm nozzles and 0.25mm nozzles it works fine. I have also run through the complete thermal calibration with a 0.8mm nozzle. Unfortunately, this does not bring any improvement.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Prusa Slicer Version 2.7.4+win64

### Operating system

Windows 10 64Bit

### Printer model

Prusa MK3s+ Firmware: e3d_REVO_fw_MK3S+_3_13_3_7094

u89djt, 2024-04-24:
(from a fellow user tracking what goes on in here)
That's really interesting. In your specific case, you can make it lay the bridge infill at 90 degrees to the solid infill by changing the "fill angle" setting to 135 degrees as a workaround. You can also rotate the object by a tiny bit and the bridge infill will flip. Maybe these changes affect where the slicer starts searching for a good direction, I don't know.
I wondered if the mesh might have caused issues because it has some strange features, but rebuilding a clean brep body of it didn't make any difference. Here's that in case it's any use to you: 
[rebuilt body.zip](https://github.com/prusa3d/PrusaSlicer/files/15097842/rebuilt.body.zip) (it's going to be handy for me too)

juckendesAuge, 2024-04-24:
Thank you @u89djt ,

my concern is that Prusa Slicer automatically offsets the bridge infill by 90° to the solid infill. If the previous setting is better for bridging, then at least the orientation of the solid infill should be adjusted.

Tupson444, 2024-04-27:
I noticed this issue recently too, although I'm still using 2.7.3.

It might have been some "bad luck" in slicing, but it was a rather simple shape (basically rectangular at the bridging infill layer), so it is more probably a bug (it shouldn't be difficult to avoid aligning the next layer when the bridging infill is all in the same direction).

dgudim, 2026-02-13:
Just stumbled on this after getting some weird artifacts on the top layer on my model, changing the infill direction helped, but yeah, I consider this a bug as well

juckendesAuge, 2026-09-18:
Over time, I've reported about 13 bugs. Many weren't addressed or even looked into at all.
Now they're being closed because a new version is out, without checking whether the problem has been fixed.
This approach teaches me only one thing.
Submitting bugs is a waste of time. It takes a lot of time and isn't appreciated at all.
The only thing you’ve achieved is that I won’t be spending any more time reporting new bugs here.
Thanks a lot for your “simple solution”!

u89djt, 2026-09-18:
We should note that the project file you originally posted also leads to the bridging and first infill layers being aligned in 3.0.0.0 alpha, so the case persists. That means it's a tough one to crack.
I can tell you that they thoroughly appreciate being given clear information about reproducible misbehaviours of their software.
You and I discussed it and may have said all the things they would have said themselves. We're a tiny, tiny speck of light in a huge constellation of effort, and it all counts.