prusa3d/PrusaSlicer #12643: PrusaSlicer was unable to automatically switch to MESA OpenGL library

Den2016, 2024-04-24:
### Description of the bug

Every time i`m trying open new slicer from menu...
Please, let prusa automatocally use --sw-renderer option

### Project file & How to reproduce

Window -> Open new instance

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

windows 7 on P8H61 board with integrated video

### Printer model

all

tempestrock, 2024-08-25:
Same on my machine: PrusaSlicer 2.8.0, Windows 10

CelvinBraun, 2025-02-21:
Might be a driver problem? I use a Nvidia card and was solving this issue, when I installed the latest driver. I installed the Game Ready Driver Package.

closed 2026-09-29 by prusaslicer-bot[bot]