prusa3d/PrusaSlicer #12648: BETTER SEAMS

Mikey600, 2024-04-25:
Ever since I first started using prusa slicer I have always thought that the seams need more control, super slicer and recently orca have addressed this with a seam gap setting. This is very handy when working with more tricky filiment and bigger extrusions. Seam gaps are common with .8 nozzles even with careful tuning , sometimes the opposite is true where the seams are too close and a not so clean joint Occur The only solution I have found is to play with restart values and acceleration which seems to help but doesn't actually fully fix the problem.... Maybe a line start acceleration setting is something to consider too affecting the start and finish of a line rather than the rest of the path🤔  maybe a little section for seam control would be nice.  Know that scarf seams are being worked on in orca slicer this looks very exciting but I think with everything said the one key thing that prusa slicer lacks is seam gap control please could we have a seam gap setting 😃😃. I hope this idea helps.

Mike 😃

BowlesCR, 2024-04-27:
Take a look at issue #11621 re: scarf joints.

Mikey600, 2024-05-08:
But this is just for normal seams scale joints are not always goings to be usable eg overhangs. Just a simple seam gap setting, similar to infill overlap setting in a way.

closed 2025-01-21 by SachCZ