prusa3d/PrusaSlicer #12651: Slicer creating a non-planer surface when laying mesh on a flat face.

jw24425, 2024-04-26:
### Description of the bug

[image]

### Project file & How to reproduce

After laying the model flat on either side, the slicer produces a non-planer surface which requires supports.

I've included a copy of the mesh file with which it exhibits this problem.

I ran it through three different slicers: PrusaSlicer 2.7.4; plus Cura 5.4.0 and ideaMaker 4.2.1.5320 and I can only get PrusaSlicer to show that strange behaviour.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows 11

### Printer model

Creality Ender 3

neophyl, 2024-04-26:
Cant see any attached mesh.  You need to zip up files for them to be a valid attachment.

jw24425, 2024-04-26:
I'm unsure of how to add it to the github bug report, but I've attached a copy of a zip file with the .3mf file on this email/post.
[12651.zip](https://github.com/prusa3d/PrusaSlicer/files/15132866/12651.zip)

closed 2026-09-29 by prusaslicer-bot[bot]