prusa3d/PrusaSlicer #12654: Big Blob of filament after M600

thedizze, 2024-04-26:
### Description of the bug

After performing a filament change with M600 a big blob of filament was dropped on the first move over the printed part 
Issue in previous version with same setting was not present or it dropped just little pipmle of filament

### Project file & How to reproduce

.

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

windows 10

### Printer model

Twotrees sapphire sp5

carlosjln, 2024-04-27:
This same thing has happened to me every time I do a filament change, I have to use the tweezers to pull the filament until the nozzle starts to move.

The pause action should ideally happen in the middle of an infill path so that any potential blobs are hidden inside the print.

rtyr, 2024-04-27:
M600 behavior is handled by the firmware not the slicer. We can check your your project file if you provide it.

carlosjln, 2024-04-27:
@rtyr thanks for the confirmation 🤔 I was suspecting that could also be the case, because it behaves similar when I tell it to pause through the screen.

Tupson444, 2024-04-27:
I think this might be related to relative/absolute E moves. If it is set different in the slicer and the firmware, the printer might extrude or retract a lot when executing M600, due to wrong E position (if absolute coordinates were e.g. 80mm, instead of a move of -1mm, it would do +79mm, or something like that).

thedizze, 2024-04-28:
> M600 behavior is handled by the firmware not the slicer. We can check your your project file if you provide it.

Hello i've added a zip with gcode an 3mf file and screenshot of my marlin m600 settings 

seems that the issue appears when I do reverse prints and I "trick" the printer like if It's a dual extruder. If I do a normal color change at layer everything is fine. 

Can you help me in understand and solve the problem ? thank you

thedizze, 2024-04-28:
[image]
what is this setting (set on 10 by default on both virtual extruders) can be influent?