prusa3d/PrusaSlicer #12656: Floating layer on wipe tower

erseln, 2024-04-26:
### Description of the bug

Floating layer on wipe tower when slicing, this is actually also printed this way.

### Project file & How to reproduce

[SpringFlat_Soft_v3.3.zip](https://github.com/prusa3d/PrusaSlicer/files/15131231/SpringFlat_Soft_v3.3.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 11

### Printer model

MK4 + MMU3

u89djt, 2024-04-26:
https://help.prusa3d.com/article/wipe-tower_125010 
[image]

rtyr, 2024-04-27:
That is expected. Please see https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)

closed 2024-04-27 by rtyr

u89djt, 2024-04-27:
@rtyr can we include a sanity check for other less experienced users like me seeing this? I initially commented with that link, but deleted the comment because of the report text, so we obviously needed you :). @erseln says "this is actually also printed this way", which implies extrusion, so perhaps the right expression - which seems obvious once it's pointed out and you've confirmed it, but it made me wonder what they meant - could be "the printer head actually follows those paths"? No filament is intended to leave the extruder during the ramming, but I guess it happens above the wipe tower in case there's leakage to avoid marring any objects?
If that's right, then the text of the explanation could include "The head is moved across the wipe tower during ramming to account for any cases where lower quality filament might lead to leakage, and ensures that this will deposit on the tower rather than objects."