prusa3d/PrusaSlicer #12657: Add Flow and Cooling Modifier/Settings for each object

Racemade, 2024-04-26:
Is your feature request related to a problem? Please describe.
Prusaslicer is one of the few slicers that still lags behind in terms of individual adjustability (especially when it comes to multiple objects). The Superslicer (albeit outdated) and especially the Orca Slicer are much better in this respect.
In addition to the latest Z-Seam functionality, Orca Slicer also offers many calibration options. Temp towers, VFA tests, flow tests etc. can be created with a simple click.
However, there are already enough requests in Github, which is why I have a small request with a lot of potential.

**Describe the solution you'd like**
To remedy this quickly, I would imagine more "modifiers", which can be added per object (See screenshot for explanation). For example, to be able to perform good calibration tests, it is worth adding filament and cooling settings.

**Describe how it would work**
Right-click on an individual object, then select "Filament" or "Cooling".

What options should be available?
Filament:
-Extrusion Multiplier
Cooling (at least):
- Fan speed (min)
- Fan speed (max)
- Activate dynamic fan speeds:
- Speed for 0% overlap
- Speed for 25% overlap 
- Speed for 50% overlap 
- Speed for 75% overlap
- etc.....

Thanks!
[image]

Garbg, 2024-05-11:
This would be especially useful using a height range modifier after a color change to change the extrusion multiplier

Racemade, 2024-06-11:
UP

Racemade, 2024-10-05:
Any chance for implementation yet?

Racemade, 2024-12-15:
Flow Test pls:)

Racemade, 2025-04-13:
Up😎

leomick, 2025-04-13:
@Racemade Please stop bumping. It's very annoying.

closed 2026-09-29 by prusaslicer-bot[bot]