prusa3d/PrusaSlicer #12658: Text Tool: MacOS on arm64: cannot add CR / LF; cannot paste from outside text editor; cannot select all text with a keystroke; cannot backspace; cannot navigate using left/right arrow keys

theschles, 2024-04-26:
### Description of the bug

I'm running MacOS Sonoma 14.4.1 arm64 (aka "Apple Silicon")

Using the Text Tool, I wanted to create multi-line text embossed onto an object: 
```
Thing
<number>
```
...as an identifier for the object being sliced / printed

Shift-Enter / Control-Enter / Option-Enter / Command-Enter -- none will introduce a carriage-return / line-feed.

I asked about it in the [Prusa3D Discord Server Help Room](https://discord.com/channels/497001479320633354/497148677219942420/1233450064156823632)

User "Blue" said in Windows he could use Shift-Enter without a problem -- and suggested writing the text in a plaintext editor and then copy/pasting it into the Text Tool dialog.

Nope, couldn't paste from Sublime Text.  It looks like copy/paste currently only works for text that's selected WITHIN PrusaSlicer.

Also, I'm unable to select-all text within the Text Tool with a Command-A / Ctrl-A / Option-A

I can't backspace over the text either -- whether I mouse-click to the right of the text in the dialog, or if I mouse-click in the middle of the text in the dialog

I also can't move the cursor within the dialog with left arrow or right arrow.

### Project file & How to reproduce

[Trying different caps.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/15133150/Trying.different.caps.3mf.zip)

Steps to reproduce:

1. Using MacOS Sonoma 14.4.1 arm64 edition
2. Download and add to empty build plate the STL for https://www.thingiverse.com/thing:1177064 -- make sure the "top" of the cap is oriented down
3. Download and add to the build plate the STL for https://www.thingiverse.com/thing:3612382 -- resizing the X/Y to 37.7 mm diameter -- make sure the "top" of the cap is oriented down
4. In Text Tool, try to add multi-line text to center of each:
* Add a CR/LF character
* Copy text from outside text editor and try to paste it into the Text Tool, both with Cmd-V or via Edit Menu > Paste
* Try to select all text with Cmd-A
* Try to left and right arrow around the text

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4+MacOS-arm64

### Operating system

MacOS Sonoma 14.4.1

### Printer model

Ender 2 Pro

Jony01, 2024-05-06:
Can't reproduce.
Would you capture screen video and upload there?
We would like to see what is happening on your computer.

theschles, 2024-06-01:
And now I can't reproduce either. ¯\_(ツ)_/¯

closed 2024-06-01 by theschles