prusa3d/PrusaSlicer #12660: Text is cut off in the macOS Installer

dstulken, 2024-04-26:
### Description of the bug

In the macOS "installer" (after opening the PrusaSlicer dmg file), there is the dialog where the user is to drag "Original Prusa Drivers" into the Applications folder.

Below this there is text, which is intended to clarify that the installation handles both x86 and ARM.  However, in the default window sizing, this text is cut off.

[image]

It appears as though there is excess space above the draggable elements, so shifting everything upward would be an easy solution.

### Project file & How to reproduce

(See above)

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

macOS Sonoma 14.4.1

### Printer model

Prusa Mini

phatfly, 2024-05-06:
Is the Prusa Slicer download suppose to install the Prusa Drivers?

ahti, 2024-09-24:
It doesn't install any drivers, it's just a folder that contains the actual slicer .app and two links to sample gcode and a new users guide. This is pretty unidiomatic. The DMG should just have the slicer .app to drag into the Applications folder.

phatfly, 2024-09-24:
Thank you,
Christopher Scott

(phone)

closed 2026-09-29 by prusaslicer-bot[bot]