prusa3d/PrusaSlicer #12664: Triple printing time (and low automatic speed) with different model orientation.

Oscar-KK, 2024-04-27:
### Description of the bug

Just an interesting fact about Prusaslicer's behavior 🙂 When placing this particular model (https://www.printables.com/cs/model/821813-charmander-flexi-articulated-pokemon-print-in-plac) parallel to the pad (as PS automatically puts it there), the calculated resulting print time is about 7:40 and the calculated print speeds are very low. However, just rotate the model by even a single degree or change the size by just one percent (without changing the position) and suddenly the time drops by five hours to about 2:40. However, it must not be rotated 90, 180 and 270 degrees, this is again at 7:40 (only resizing will help there), any other rotation is at 2:40. All this with the same settings (speeds are calculated automatically - max, speed set to 100). I haven't noticed this behavior in other models yet.
[image]
[image]

### Project file & How to reproduce

[charmanders.zip](https://github.com/prusa3d/PrusaSlicer/files/15139104/charmanders.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4.

### Operating system

Windows 10

### Printer model

Creality Ender 3 S1 + Sonic Pad (Klipper)

Tupson444, 2024-04-27:
I've also noticed some inconsistencies in print time, although not as drastic as this example. I first thought that time estimate just isn't reliable, but I tested the files sliced with different time estimates, and they actually printed differently (very close to the estimates), so there is some error in the generated GCODE that unnecessarily makes the print time longer (with the same slicer settings and same STL).

u89djt, 2024-04-28:
Maybe you could include the 3mf file of it at the alignment that causes the long print time. I had to rotate the one you included, but it did work. 
[image]

murk-sy, 2024-05-03:
I made the issue reproducible with these 2 layers of a bit of the tail. You can most easily check if the bug is reproduced by enabling speed legend and seeing speeds of ~13. Should be much easier and faster to reproduce now.
[image]
Note that the issue does not appear if you make the cutoffs even slightly narrower from any side.

Feature:
[image]

Volumetric flow rate:
[image]

What I suspect is the culprit:
[image]

### IMPORTANT: If the model is cut, even if the issue stays reproducible, it may not work after the project is reopened!
Just to spare you some of my own confusion.

Project and sliced gcode in case it's somehow not reproducible again
[2024-05-03 Prusaslicer issue 12664 minimised.zip](https://github.com/prusa3d/PrusaSlicer/files/15198437/2024-05-03.Prusaslicer.issue.12664.minimised.zip)

closed 2026-09-29 by prusaslicer-bot[bot]