prusa3d/PrusaSlicer #12666: Vase mode glitch

MxTale, 2024-04-28:
### Description of the bug

When slicing in vase mode, the GCODE generated makes the nozzle go trought the last layer
[image]

### Project file & How to reproduce

[image]
[PlantPot_DiamondCut.zip](https://github.com/prusa3d/PrusaSlicer/files/15142425/PlantPot_DiamondCut.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

W11

### Printer model

Anycubic Kobra

u89djt, 2024-04-28:
Your model has spikes at the top, so there's no sensible way to print the last one-and-a-bit layers.
[image]
[image]
[image]
When you cut the spikes off, it slices ok.
[image]
(not affiliated with Prusa)

MxTale, 2024-04-29:
Thanks Dave!

image-1.png (view on web)<https://github.com/prusa3d/PrusaSlicer/assets/75209005/27437b9c-a265-4668-917b-b377e36c92c4>
image.png (view on web)<https://github.com/prusa3d/PrusaSlicer/assets/75209005/04b718d1-24da-4ef2-9819-92a5294f3df4>
image-2.png (view on web)<https://github.com/prusa3d/PrusaSlicer/assets/75209005/b700f346-f06c-493f-a74f-65013e608a67>
image-1.png (view on web)<https://github.com/prusa3d/PrusaSlicer/assets/75209005/66b9f794-0a65-4074-8628-71887705cd27>

u89djt, 2024-04-29:
@MxTale You're very welcome! I always play around with models on github issues in case there's something I haven't seen yet.  I had a go at making it print the whole volume in vase mode out of interest. I think there's no point, since it takes longer than a normal print, but you might enjoy seeing it: 
[spiralled base PlantPot_DiamondCut.zip](https://github.com/prusa3d/PrusaSlicer/files/15148942/spiralled.base.PlantPot_DiamondCut.zip)

closed 2026-09-29 by prusaslicer-bot[bot]