prusa3d/PrusaSlicer #12668: Organic supports generated in mid air

CanuckPlay, 2024-04-29:
### Description of the bug

Often, when organic supports are generated, portions are created hanging in mid air.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2+win64

### Operating system

Windows 10.

### Printer model

Anycubic Kobra 2 Max

neophyl, 2024-04-29:
Could you please search for existing issues before opening new ones.  This has been reported so many times now.  The primary one is https://github.com/prusa3d/PrusaSlicer/issues/9601  opened in Feb last year.

CanuckPlay, 2024-04-29:
> This has been reported so many times now. The primary one is #9601 [1]

Not really my job to chase down what bugs have and have not been 
reported. If that bug has existed for over a year, maybe they should 
just fix it.

Links:
------
[1] https://github.com/prusa3d/PrusaSlicer/issues/9601
[2] 
https://github.com/prusa3d/PrusaSlicer/issues/12668#issuecomment-2081955558
[3] 
https://github.com/

closed 2026-09-29 by prusaslicer-bot[bot]