prusa3d/PrusaSlicer #12671: Error Prusaslicer

YuriuTrosyn, 2024-04-30:
### Description of the bug

After slicing prusaslicer show error and closing
[Error log.txt](https://github.com/prusa3d/PrusaSlicer/files/15168781/Error.log.txt)

### Project file & How to reproduce

[Cup.zip](https://github.com/prusa3d/PrusaSlicer/files/15168839/Cup.zip)

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

7.4.2

### Operating system

Windows 10

### Printer model

MK4

murk-sy, 2024-05-03:
Can't reproduce on my machine, slices fine

Possibly duplicate of https://github.com/prusa3d/PrusaSlicer/issues/12387 https://github.com/prusa3d/PrusaSlicer/issues/12236
Can't reproduce these either

Can you go to "Help>System Info>Copy to clipboard" and post that? Might help devs narrow in on the issue.
Does it happen with other models? Does it always happen with that model or only occasionally?

YuriuTrosyn, 2024-05-03:
PrusaSlicer
Version:   2.7.4+win64
Build:     PrusaSlicer-2.7.4+win64-202404050928

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 17,047MB
OpenGL installation
GL version:   4.6.0 (4.6.0 - Build 31.0.101.2111)
Profile:      Core
Vendor:       Intel
Renderer:     Intel(R) HD Graphics 630
GLSL version: 4.60.0
Textures compression:       Enabled
<details>
<summary>Installed extensions:</summary>
GL_3DFX_texture_compression_FXT1
GL_AMD_depth_clamp_separate
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_cl_event
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
[... 166 lines cut ...]
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_shader_subgroup
GL_KHR_shader_subgroup_arithmetic
GL_KHR_shader_subgroup_ballot
GL_KHR_shader_subgroup_basic
GL_KHR_shader_subgroup_clustered
GL_KHR_shader_subgroup_quad
GL_KHR_shader_subgroup_shuffle
GL_KHR_shader_subgroup_shuffle_relative
GL_KHR_shader_subgroup_vote
GL_KHR_texture_compression_astc_hdr
GL_KHR_texture_compression_astc_ldr
GL_NV_blend_square
GL_NV_conditional_render
GL_NV_primitive_restart
GL_NV_texgen_reflection
GL_SGIS_generate_mipmap
GL_SGIS_texture_edge_clamp
GL_SGIS_texture_lod
GL_SUN_multi_draw_arrays
GL_WIN_swap_hint
WGL_EXT_swap_control
</details>

YuriuTrosyn, 2024-05-03:
[image]

YuriuTrosyn, 2024-05-03:
I have problem only this project.

YuriuTrosyn, 2024-05-05:
I have problem only this poject

closed 2026-09-29 by prusaslicer-bot[bot]