prusa3d/PrusaSlicer #12672: extremely slow print speed with Vase mode in 2.7.4 but not 2.7.2

BloodyRain2k, 2024-04-30:
### Description of the bug

I was trying to print a part in vase mode and got wondering why that tiny thing would require **1 1/2 h**.
So I checked the print speed and 2.7.4 decided for some reason that it want to print the entire vase section after the base at **0.1 mm/s**.
[image]

I have no idea how this even happened, but the setting "min. speed" is definitely being ignored.
I also checked with 2.7.2, because I was sure that this issue never happened before and indeed, in 2.7.2 it would print in 4 min.

I've already messed with all kinds of settings, turning on and off auto fan speed and layer speed, nothing seems to convince 2.7.4 to change it's mind about the 0.1 mm/s.

### Project file & How to reproduce

[Hose_Parts-8mm_Nozzle.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/15170948/Hose_Parts-8mm_Nozzle.3mf.zip)

Loaded this part into the slicer, turned on vase mode, found the issue.

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

PrusaSlicer-2.7.4+linux-x64-GTK3-202404050928

### Operating system

Debian GNU/Linux n/a (13?) trixie (x86-64)

### Printer model

CR-6 SE

jmd73, 2024-05-01:
@BloodyRain2k Hello!
All I can see is that setting Speed->Pressure equalizer (experimental) values to 0 instead of 0.8 gets you 4 minutes.
[image]

Maybe there is a difference between those settings between the versions?

BloodyRain2k, 2024-05-01:
Good find, those might be the cause. Increasing them to 2 changes the printing time to 43 min.
But 2.7.2 has the exact same settings and no issue, so these seem to be in some formula that got changed and now breaks.

Something that I noticed this time looking at the various layers is this:
[image]
I have to increase the setting from 0.8 to 5 to even get a tiny gradient (not shown here) where 0.8 in 2.7.2 (right one) gets me a large gradient.

closed 2026-09-29 by prusaslicer-bot[bot]