prusa3d/PrusaSlicer #12677: Organic supports are generated outside of build plate

vrosu, 2024-05-03:
### Description of the bug

I'm trying to use organic supports to print a part that's unfortunately only printable like this (due to a length of 240 mm).
When activating organic supports I observe that some of them are generated outside of the build plate, even if **apparently** there is enough space in the object itself.
I would expect that the organic supports are generated only in the build plate area (on the build plate or in the object, but not outside build plate boundaries).
I haven't been able to locate that would force the system to generate organic supports only inside the object. 
Just in case, snug supports are generated correctly, but I'm trying to avoid spending so much filament and time.
[image]

### Project file & How to reproduce

[Support.zip](https://github.com/prusa3d/PrusaSlicer/files/15199476/Support.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 11 Home

### Printer model

MK4

murk-sy, 2024-05-03:
### Workaround
Since support blockers don't do much, what you can do is place very thin (0.05 mm) parts that cannot be printed right next to the part.

[image]

[image]

[Support-with-workaround.zip](https://github.com/prusa3d/PrusaSlicer/files/15199958/Support-with-workaround.zip)

**Side note:** if the "blocker" part is slightly too far from the base part, Prusaslicer will still try to place a tree support on the build plate:
[image]
More like twig supports, what am I even looking at

vrosu, 2024-05-03:
Ha, it works indeed @murk-sy - thanks. I tried replicating your project via the instructions first, and there was one crucial thing missing - the fact that the objects needed to be merged - which I observed when looking at the project object tree in the right side of the page.
Much appreciated!
Interesting that with organic supports I get 250g of plastic less (positive), but increase of 14 hours (negative). 
I will try @u89djt 's suggestion - thanks for that as well. The reason I did not try that yet is because always it's a continuous exercise to try and rotate in whatever axis the object and see if it fits + painting custom supports hoping it will work.... Final material is PLA, I set PETG because I thought I have enough filament in my PETG roll, but apparently it's not enough.

(I would still keep this bug report open, as I'd expect Slicer not to generated them on the outside.)

vrosu, 2024-05-03:
Correct, but I'd like official prusa then to close with duplicate, as this needs more attention (I might need to look into this myself)

u89djt, 2024-05-03:
and it'll be handy for other folk looking to report the same issue to find @murk-sy 's fix. I guess it's a grey area with workarounds - maybe there could be a "workaround" category between open and closed?

Jan-Soustruznik, 2024-09-17:
Supports that are generated out of bed in the known issue. https://github.com/prusa3d/PrusaSlicer/issues/9569 and original https://github.com/prusa3d/PrusaSlicer/issues/849

closed 2024-09-17 by Jan-Soustruznik