prusa3d/PrusaSlicer #12681: Splitting multiple, hollow STLs (that you have earlier Merged), gives insane results.

S3D-r, 2024-05-04:
### Description of the bug

If you first merge and then split two _hollow_ STLs, after spitting you'll have four solid items. I.e. the internal "antimaterial" (cavity) will turn into material and the original, hollow item is no longer hollow but solid. 

Expected result: After Splitting you have two hollow items, just like before Merging.

### Project file & How to reproduce

Easy to test: Make 2 hollow cubes. Merge these. Split them. Voilà!
For some reason I have the "Split to Parts" grayed out, so that th

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.0.1

### Operating system

Win10-64

### Printer model

MK3S+

u89djt, 2024-05-04:
Hi, you didn't finish putting together the bug report:
[image]

S3D-r, 2024-05-04:
Sorry, here it is.
Also, the S/W version is 2.7.2 (not 2.0.1)
[Merged.zip](https://github.com/prusa3d/PrusaSlicer/files/15210856/Merged.zip)

u89djt, 2024-05-04:
(just to be clear - I'm a fellow user just taking an interest in issues as they go by)
Can confirm that the split action results in four object in version 2.7.4 for me.
If I right click each instance within the merged group and select "fix by windows repair algorithm", then split, I get two hollow cubes.
I've gone ahead and created a hollow cube in fusion, exported that to the slicer, copied and pasted, merged and split and got the same result as you - four objects. If I repair the meshes before or after merging, and follow that with splitting, I get two objects.
So, problem reproduced, and its seems to be an issue with the slicer wanting the meshes to have different characteristics to what comes out of your and my ways of making them. So here's my version before merging and not repaired in case they add anything:
[fusion cubes merged.zip](https://github.com/prusa3d/PrusaSlicer/files/15210918/fusion.cubes.merged.zip)

closed 2026-09-29 by prusaslicer-bot[bot]