prusa3d/PrusaSlicer #12683: When I save a set of print settings, very few settings are actually saved. When I open Prusa Slicer for a new project, I have to go and reset all of the same settings over and over and over.  I would LOVE a preference page that allows me to set USER DEFAULTS.

AlaskainFlight, 2024-05-06:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe how it would work**
Be as descriptive as your skill level allows you.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or paint drawing about the feature request here.

m-cas, 2024-05-07:
while this feature request gets evaluated, may I suggest you create an empty project with all the settings you need and use this to start all your new projects by importing the stl files (or whatever you start with) and then shift-saving, leaving the original project file ready for the next project?

AlaskainFlight, 2024-05-16:
> Could you elaborate which settings aren't saved and what your workflow is? I'm not exactly sure if you're talking about print settings presets or something else.
>
> If possible, provide a step by step guide that reproduces the issue reliably.

Hi Murk,

 

I’ll give you the scenario that baffles me the most. 

I save a project as a 3MF and the print setting into a modified copy. When I reopen the project (or the settings) these parameters have went back to someone’s default settings, not mine and not what the slicer says the default should be on some of these parameters!

 

*	Perimeters resets to 2
*	Solid layers (Top) resets to 6, (Bottom) resets to 5
*	Minimum shell thickness (Top) resets to .7, (Bottom) resets to .5
*	Avoid crossing perimeters resets to off
*	Avoid crossing perimeters – Max detour length resets to off
*	Infill resets to 15% Grid
*	Combine infill every: resets to 1
*	Brim resets to Outer brim only
*	Perimeter, Infill, and Solid infill extruders reset to 1
*	Infill/perimeters overlap resets to 15%

 

It’s even more of a pain in the ass if I switch from the MK4 back to the XL, then I have to go back and tell it which filaments were there when the 3MF was saved!!

 

In my opinion the slicer should either save the last setting I used (Cura) or allow me to set my own default parameters and save them.

 

Dave West

closed 2026-09-29 by prusaslicer-bot[bot]