prusa3d/PrusaSlicer #12686: PrusaSlicer Crashing with the attached project

samc99us, 2024-05-06:
### Description of the bug

Prusaslicer crashes while slicing the attached project file. 
[Tacktick Mount Top 38mm Pole 1.8mm Wall HeatSet Inserts v2_PrusaSlicer.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/15223703/Tacktick.Mount.Top.38mm.Pole.1.8mm.Wall.HeatSet.Inserts.v2_PrusaSlicer.3mf.zip)

### Project file & How to reproduce

Attempt to slice the part file as-is.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows 11

### Printer model

Voron 2.4, 350 mm3, with High Flow Stealthburner

samc99us, 2024-05-06:
Update: downgraded to PrusaSlicer 2.7.3, it crashes there as well.

Jan-Soustruznik, 2024-05-10:
Hello, @samc99us , thank you for you report. We will try to fix in some next release. Issue created under SPE-2298, thank you, have a nice day

maxcom99, 2024-05-17:
In case relevant or perhaps it helps someone in the future- I was trying to slice 25 parts on a single plate for upgrading my MK3S+ to MK4 (Printing them on my additional MK4 is nearly twice the speed - 22h vs 36h using pre-sliced MK3 gcodes)  Anyway, after about 6 parts, the slicer would crash when trying to slice.

My work-around was to lay out the parts, make them all disabled/not printable. Enable "Background Processing" under Configuration->Preferences and then enable one part at a time while waiting for each part to finish slicing before enabling the next. Perhaps worth exploring the multithreading or GPU hooks?

This is a top of the line desktop basically so horsepower is not an issue. 32core i9 w/196Gb DDR5 Nvidia 4090 etc. Latest drivers, Win 11 Enterprise w/ 274 slicer and updated profiles. I don't mean to piggyback on this report so if my findings are different / unrelated please let me know and I will open a new report with requested info.

andrewluebke, 2024-05-23:
This crashes on my computer as well.

lukasmatena, 2024-06-21:
Should be fixed in [2.8.0-beta1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.8.0-beta1). Closing.

maxcom99, 2024-06-22:
[rick_multicolor.zip](https://github.com/user-attachments/files/15934543/rick_multicolor.zip)

2.8.0-b1 crashes slicing this still as before

murk-sy, 2025-07-30:
Both Tacktick.Mount and rick_multicolor files slice fine in 2.9.2, can probably be marked as resolved

closed 2026-09-29 by prusaslicer-bot[bot]