prusa3d/PrusaSlicer #12688: Add Rectilinear Interlaced Interface Layers

morphias2004, 2024-05-07:
Please add the option for 'Rectiliner Interlaced' to the setting for Interface Layers.

This results in smoother top layers for the interface materials as a result of the layers printing at 90 degrees to each other. This further results in a smoother bottom lay on the model.

It also makes the interface layers more reliable as they are not direct stacking on top of each other in the same orientation.

It's available in Orca Slicer.

In addition to this, the first interface layer needs to be in the same material as the rest of the supports to make the first interface layer in the dissimilar material much more reliable. This is essentially covered in case #12062 

https://github.com/prusa3d/PrusaSlicer/issues/12062

It's available in Orca Slicer as well.

Prusa Slicer Options:

[image]

Orca Slicer Options:

[image]

Prusa Slicer Interface Layers:

[image]

Orca Sliver Interface Layers with Rectilinear Interlaced:

[image]

Prusa Slicer dissimilar interface interface:

[image]

Orca Slicer dissimilar interface interface:

[image]

argosforge, 2024-05-07:
I said this on the other thread and I'll say it here too:

This change is critical to making multi-material supports work - it just simply DOES NOT work (adhere) otherwise.
We need this to take full advantage of the XL multi-material handling capabilities

morphias2004, 2024-07-20:
Come on guys. This is a simple thing to implement as it is already in derivative slicers.

This issue and #12062 are essential to have excellent support interface layers.

morphias2004, 2024-08-16:
Still waiting.

Hausmeister486, 2024-12-28:
Suggested solution for the Prusa Slicer:
Flexible selection and settings of support structures and interface layers in terms of the number of layers and which material or extruder is used for each layer.
The whole thing separately for the top and bottom of the support.
A global bed temperature setting even when using different temperatures of the support materials.

Hausmeister486, 2026-09-18:
Over two years on and this still hasn't been implemented. :(

morphias2004, 2026-09-18:
> Over two years on and this still hasn't been implemented. :(

Absolute joke hey?!

I love how they've just closed all the open cases because PS3 is coming out. Initial testing shows most of these issues have not been addressed.