prusa3d/PrusaSlicer #12694: Issue when using colorchange

WhoTheHellIsAlice, 2024-05-08:
### Description of the bug

There seems to be an issue in the slicer when using the setting Multiple extruders with only a single extruder installed.
I've been playing around with printing multiple colors in a single layer, and I have been noticing a behavior which I can reproduce. At first I thought there was dirt in my nozzle or extruder, which means I have dots with the wrong colors in my prints. I've uploaded a picture of to different prints with different filaments and print surfaces, and the dots are exactly the same at the same spot in both prints. Notice the two orange dots in the Y and the black dot in the left orange arrow.
What I've seen, when observing the printer working, is that after a color change, the nozzle returns to the position it left from to go for the color change, and it then prints a dot of the new color in that position, before it moves to where it should actually start printing the new color.

The printing order of the first layer was:
1. White
2. Orange
3. Black

It also extrudes a big dot on the edge next to the Y. Again same procedure on both prints.

[image]

### Project file & How to reproduce

[GPS cover.zip](https://github.com/prusa3d/PrusaSlicer/files/15247694/GPS.cover.zip)

I've attached the gcode, and the printer will make the same behavior everytime it's being printed.

[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+

### Operating system

Windows 10

### Printer model

Prusa Mk4

Unostot, 2024-06-12:
I actually observed a similar behaviour a few days ago, but on a XL with 5 heads:
I was printing multiple objects, each with its own color. Thus I did not use the wipe tower. 
I'm not sure right now, but perhaps can look at the gcode at home later, but the behaviour was:
It was:
* the tool was finished with its object, goes to the next object (which should have a different color and thus toolhead), from there to parking, switching tool -> printing next object
* Or it was going to park. Next toolhead goes to the old object, left a tiny bit filament there, and goes to the correct object...

As I said ,not sure which version it did...but the result was a lot of small string and of course small blobs of wrong filament.
For the print in question it was no real issue, but for mor 'important' prints this is a behaviour which is not really nice...

When using the wipe tower, this problem does not occur, since then everything ends up there. But for printing multiple objects with different tools, each tool another object...The wipe tower would not be needed, but since the tools then go for a moment to the wrong object, this leads to unnecessary 'pollution' and in case ob e.g. PETG and PLA not only color contamination but weak structural integrety as well, because of non sticking residues here and there...

Situation was:
* different objects, each with a personal extruder
* no wipe tower
* I did take care that the objects are next to their extruder, but this did not help

dmbaio, 2024-09-15:
It is the same issue as #9036, because the logic previously being applied to the color change is also applied to tool changes.

cc: @Nohus who fixed this issue for color changes

Nohus, 2024-09-15:
> cc: @Nohus who fixed this issue for color changes with PR #9036.

@alchitry already opened a PR replicating my color change fix for tool changes: #13347

rbuehlma, 2024-10-20:
I use the following "Tool change G-code" as a workaround:
`
G91

G1 Z3 F720

G92 E0

G90

M83

M600
`

This lifts the nozzle prior moving the the color change by 3mm. After the color change, it moves back to the same position (again 3mm above the object) and then moves to the first extrusion point for the new color. This way, the new color does not get dropped at the old location.

closed 2026-09-29 by prusaslicer-bot[bot]