prusa3d/PrusaSlicer #12697: SVG Depth precision

gaby64, 2024-05-08:
### Description of the bug

0.13mm gets rounded down to 0.1mm
it is correctly applied but when I edit im never sure if its using 0.13mm or 0.1mm

### Project file & How to reproduce

N/A

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Windows 10

### Printer model

two trees sapphire plus

murk-sy, 2024-05-09:
+1, just an inconsistency

Text on the other hand allows for 2 decimals for depth - though I think height should also just be allowed to be entered to 2 decimals if anyone really needs to.
[image]

gaby64, 2024-05-09:
I would like 3 decimals

closed 2026-09-29 by prusaslicer-bot[bot]