prusa3d/PrusaSlicer #12702: Multiple bottom layers on wipe tower to improve build plate adhesion on tall prints with the prusa XL

MaximilianBuegler, 2024-05-10:
**Is your feature request related to a problem? Please describe.**
My wipe tower came off the build plate multiple times during very tall prints on my Prusa XL, even when printing with a massive brim.

**Describe the solution you'd like**
I believe having multiple bottom layers on the wipe tower (or the brim around it) would increase adhesion to the build plate preventing my tall prints from failing. 

**Describe how it would work**
Adding a parameter to specify the number of bottom layers for the wipe tower

**Describe alternatives you've considered**
I tried increasing the brim and line spacing of the wipe tower, but the single bottom layer just doesn't seem enough when the tower gets > 20cm tall.

**Additional context**

morphias2004, 2024-05-10:
Changing the first layer density, interface pattern spacing and the support extrusion width will resolve this in most instances.

Increase the extrusion width to match the nozzle diameter or just (0.02) under. By default, the extrusion width is too narrow.

[image]

[image]

dwineman, 2024-06-14:
If it has to purge on every layer of your print, it can't do this. It will always print the tower with solid perimeters, but when purging has to happen on layer 1, the first layer of the tower will be made of purge lines and not the nicely adhering, carefully printed solid base that you want.

It could potentially ensure a single solid bottom tower layer by doing first-layer purges elsewhere. The initial purge line is printed along the edge of the plate and takes up half the width, so there's enough room for 8 such lines around the border of the plate. That's plenty because the first layer can have at most 4 filament changes (plus the initial purge). So that's a strategy that might work, assuming that amount of purge is sufficient. But I don't think it would be practical to have *multiple* solid bottom layers unless there are no color changes on those layers.

Instead, try turning on the stability cone in the wipe tower settings.

pipcorona, 2024-10-02:
I would like this feature for a different reason. I have several different prints where the tool changes aren't until higher up in the prints. So I get a nice first layer, and then several sparse layers. I primarily print out of PETG so when I go to remove the prime tower, the first layer breaks apart and leaves this, .2mm pieces on the bed. Enforcing a certain number of filled layers for the first X number of layers would help keep it together while removing it.

I imagine there could be an option like "Enforce full layers on prime tower" and then a number of layers. That way if the prime tower isn't filled with priming, it would fill it after each layer, instead of reverting to the sparse infil it is doing when it doesn't switch tools.

morphias2004, 2026-09-18:
Over 2 years, no action and just gets closed. Still an issue in PS 2.9.6. Is this fixed in PS3?