prusa3d/PrusaSlicer #12703: Perimeter Printing Order feature be merged back from OrcaSlicer

5jvm0u4, 2024-05-10:
As my previous feature request stated, I'm printing lots of PACF with 4 or more perimeter, and the size of small holes became off.

So I was forced to use OrcaSlicer to take adventage of this feature from it, but it's not as friendly to use as our native PrusaSlicer, and also a bit buggy with path generation.

But the feature does work and the option inner/outer/inner gave me not only good dimantional accrucy but also good surface finish, rather then the only option in PS (External perimeter first) gave me bad surface finish.

5jvm0u4, 2026-09-03:
Up

Jan-Soustruznik, 2026-10-20:
Hello, A similar feature request already exists. Closed ticked as duplicate of #9393. thank you, have a nice day

closed 2026-10-20 by Jan-Soustruznik