prusa3d/PrusaSlicer #12704: manque des parois

18pool49, 2024-05-10:
### Description of the bug

dans un vase  composé de 22 morceaux >> je demande 2 ou 4 parois MAIS 
il y a QU'UNE ( 1 ) PAROIS  seulement de réalisée a l'écran 

### Project file & How to reproduce

[Dandelion 4par_A+B+c+d+e+f+g+h projet pr.zip](https://github.com/prusa3d/PrusaSlicer/files/15273675/Dandelion.4par_A%2BB%2Bc%2Bd%2Be%2Bf%2Bg%2Bh.projet.pr.zip)
[image]
[image]
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

W11 

### Printer model

geeetech Mizar M

neophyl, 2024-05-10:
I don't think you quite understand how the slicer works.  It doesn't matter how many perimeters you ask for if the model isn't thick enough to fit them in.  You could ask for 100 and if theres only room for 60 then you get 60.

Your model has a wall thickness of 0.378mm (at the rim).  Your printer has nozzles defined as being 0.44mm in size (an unusual size btw) and your extrusion widths are set to 0.45mm (normal for a 0.4mm nozzle profile).  So your wall thickness is actually less than your extrusion width.  
Its quite obvious that it can only fit a single perimeter into that space.  And its only through Arachne adjusting the extruded width dynamically that its giving you ANY perimeters at all.  In fact there are gaps where the wall thickness goes too thin even for the arachne settings to slice a perimeter.
[image]

If you want more perimeters then make your models walls thicker.

If you don't have the ability to modify the model then you are left with something like adding an xy compensation (Advanced>XY size compensation) to make the  model thicker on a per layer basis.  That will expand all the perimeters outwards by the configured value, so the exterior will grow and the inside of the vase will also grow, reducing the space inside.  0.6mm (so making it thicker by 1.2mm overall) is needed for 4 perimeters.  
Of course this being a multi colour model that messes up the separate parts as they overlap so that's not going to give you the result you probably want.  

That leaves you will editing the original model to have a thicker shell.

As such theres no software bug here.  It's giving you what it can within your designs constraints.

18pool49, 2024-05-10:
un grand merci ; pour vos explications qui mettent en avant mes incompétences 
et je comprend très bien pourquoi les STL de vase sont obstrué sur le dessus > pour nous donner un volume fermé que l'on pourra garnir de plusieurs parois >> ce qui est pas le cas pour celui ci 
la compensation X/Y fonctionne très bien 
merci a vous 
[image]

neophyl, 2024-05-10:
To correct a misconception about solid bodies and using Vase mode, it wont provide multiple walls either.  

Vase mode slicing a solid body (as opposed to slicing a vase normally) is designed for a single continuous extrusion.  As such it too produces only a single perimeter.  The thickness is controlled by your configured extrusion width.  

There are some design techniques you can use to make the 'walls' thicker like using thin slots but into the surface of the solid body so the perimeter zig zags along but that's beyond the scope of this problem.  Similar to this https://www.printables.com/model/380721-divider-vase-mode or https://www.thingiverse.com/thing:2277253 for example.

May be a language thing with Google translate (sorry for my lack of foreign language skills)  but in English I would not translate it as "incompetence", merely a lack of knowledge which can easily be remedied :)

18pool49, 2024-05-11:
merci encore "néophyle" pour vos précieux conseils 
le résultat est a la hauteur de mes espoirs 
mais je voulais savoir comment corriger une erreur de calcul de temps de dépose qui est donné par Prusa a 8h18
pour un temps réel de 10h45 
147 minutes d'écart pour les 324 changements d'outil qui doivent etre la cause de cet écart 
quel paramètre de Prusa est a modifier  pour améliorer le résultat ( 27 secondes par outil) ????
[image]

18pool49, 2024-05-11:
RE BONJOUR 
j'ai trouvé le para de temps de chargement et déchargement du fil  et le compte est bon 
[image]
[image]

closed 2024-05-11 by 18pool49