prusa3d/PrusaSlicer #12715: Rotation of camera should be around the point on the bed instead of its center

Akxe, 2024-05-14:
**Is your feature request related to a problem? Please describe.**
When an object is in the corner of the bed (as it is by default). Zooming in and then rotating to see if everything is correctly sliced is impossible, as the rotation point is around the centre of the bed

**Describe the solution you'd like**
I would like the rotation to be around the thing we want it to, usually the mouse target.

**Describe how it would work**
Project a point from the mouse to the bed and rotate around it instead. It could even project point on the model if one is present, therefor rotating nicely around the object in question.

**Describe alternatives you've considered**
Always move the part to the centre of the bed to have the same effect. After inspection, move it back (to avoid heating multiple bed zones).

**Additional context**
The camera should ideally resemble the feel of the camera in Fusion, Blender, or other 3D software...

neophyl, 2024-05-14:
Please check for existing issues before opening new ones.  This is a duplicate of several.  For example https://github.com/prusa3d/PrusaSlicer/issues/9083 or https://github.com/prusa3d/PrusaSlicer/issues/12323  

One thing you can do as mentioned in https://github.com/prusa3d/PrusaSlicer/issues/12323 is selecting the object and press 'z' key.  That will centre on the object so you can rotate around it.
You can also manually change the point by pressing and holding the right mouse button (or centre mouse button if you have one).  Rotating then goes around that.  It isn't as precise as zooming to an object though but you do get used to it.

You might want to review the list of shortcuts (Help>Keyboard shortcuts or just ?).

Akxe, 2024-05-14:
Did look for it, yet did not find it. The workaround is OK, but the actual solution would be great :/

closed 2024-05-14 by Akxe