prusa3d/PrusaSlicer #12716: Support both paint-on-color and extruder-change-at-layer in the same print job.

krhackbarth, 2024-05-14:
**Is your feature request related to a problem? Please describe.**
I’ve used paint-on color to add the color from extruder 2 at the bottom of the engraved “can”.
I now want to change to extruder 2 for all printing at Z height 10.2 (layer 51) and above. That way I don’t have to go through the trouble of painting on color for all facets of the graphic shape starting to print at this layer.
Imagine wanting to print 30 of these objects – each with different engraved text and graphic shapes – as a single print job. Note that the graphic shapes all start printing at the same layer height, so changing extruders at this height would work well for all objects.
It’s pretty easy to use paint-on-on color for the engraved text but it can be very painful to do that for the graphic shapes – some of which are very complex.
The attached 3MF file is for just one object. I want to print 30 or more of them in the same print job.
Here's the image of the object in the slicer and the associated 3mf file:
https://1drv.ms/u/s!AiUs0E-j-6ZSl84_3w3b4rXdqqRCrA?e=t5wOmU

**Describe the solution you'd like**
Support both paint-on-color and extruder-change-at-layer in the same print job.

**Describe how it would work**
described in the paragraphs above

**Describe alternatives you've considered**
As it is now I have to do all color changing through paint-on-color which can take a long time with lots of raised graphics in different shapes.  Fortunately, all raised graphics begin at the same layer for all objects.

**Additional context**
Add any other context or paint drawing about the feature request here.
none

fscheps, 2024-05-15:
I was trying to do exactly this today! I was trying to print a model which starts with one color, then has a layer change and I wanted to use the second extruder for that color change but it doesn't seem to be possible :(

krhackbarth, 2024-05-19:
> I was trying to do exactly this today! I was trying to print a model which starts with one color, then has a layer change and I wanted to use the second extruder for that color change but it doesn't seem to be possible :(

I may be misunderstanding what you're saying but I can, and have, changed to the other extruder at a layer.  You may need to update your Prusa Slicer software.

closed 2026-09-29 by prusaslicer-bot[bot]