prusa3d/PrusaSlicer #12718: External perimeters do not match the model with Arachne generator causing a layer shift and Classic generator doesn't work at all

Auxx, 2024-05-15:
### Description of the bug

Arachne generator stopped slicing external perimeters correctly in recent versions (2.7.x). The position of the external perimeter depends on internal perimeters and infill causing layer shift like artifacts. And Classic generator is completely broken and can't slice more than 7 layers even for the simpliest models.

### Project file & How to reproduce

**Steps to reproduce:**

1. Download the archive.
2. Load the STL file into Prusa Slicer.
3. Slice with Arachne generator.
4. Slice with Classic generator.

Arachne generator result:

[image]

Classic generator result:

[image]

Archive with Arachne project and Classic project 3MF files, as well as original STL and screenshots.

[Perimeter Bug.zip](https://github.com/prusa3d/PrusaSlicer/files/15320594/Perimeter.Bug.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 11

### Printer model

Creality Ender 3 S1 Pro

neophyl, 2024-05-15:
On the classic perimeter 3mf you have your extrusion width defined as 0.63 for external perimeters.  Your walls are 0.44mm thick on the model.  Hence Classic will not slice them as they are too thin to be printed.  Unless you also enable Detect thin walls, and that will basically ignore that fact and print them with a bigger thickness (0.5mm it looks like).
So no software bug there.  Doing exactly what its supposed to do.

Arachne is having similar issues.  At the lower layers its printing the external perimeters at 0.63 and a single Internal perimeter at what looks like something between 0.8 and 0.9mm wide.  
[image]

after that it is also transitioning to a thinner 0.57mm width extrusion.  It looks like the position of the centre line of the extrusion is correct but due to the different extrusion width the wall appears stepped out.
[image]

Basically all the issues are from trying to print geometry that is too thin for your chosen nozzle size.  When it can only lay down a single perimeter it can only do so much.  When there are multiple perimeters (like on the lower layers) it has more freedom to adjust things.

Auxx, 2024-05-15:
I see, you're correct. Switching to 0.4mm nozzle solves the issue. Sorry for a user mistake and thanks for the explanations.

closed 2024-05-15 by Auxx