prusa3d/PrusaSlicer #12720: Fix by Windows repair algorithm

lexcomeing, 2024-05-16:
### Description of the bug

I compiled the PrusaSlicer-2.7.4 source code using Visual Studio 2022, and after a successful compilation, I attempted to repair a part but found that there is no "Fix by Windows repair algorithm" button. My operating system is Windows 10.
But the official one has a "Fix by Windows repair algorithm" button.

### Project file & How to reproduce

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.2

### Operating system

Windows 11 专业版

### Printer model

NULL

lukasmatena, 2024-05-16:
You must install Win10 SDK and it has to be found by CMake at configure time. Search for `WIN10SDK_PATH` in the root CMakeLists.txt file. Closing.

closed 2024-05-16 by lukasmatena

lexcomeing, 2024-05-17:
Thank you for your help

> You must install Win10 SDK and it has to be found by CMake at configure time. Search for `WIN10SDK_PATH` in the root CMakeLists.txt file. Closing.

Thank you for your help!I'll give it a try