prusa3d/PrusaSlicer #12721: Adopt "No perimeters on bridging areas" setting from SuperSlicer.

thisloginalsotaken, 2024-05-16:
**Is your feature request related to a problem? Please describe.**
Problem: when bridge is supported by thin wall, thinner than perimiter count from settings, perimeters a printed in the air, and bridging fails
[image]
"Detect bridging perimeters setting" makes things even worse, removing the bridge completely.
[image]
Here are .stl and .m3f files illustrating the problem.
[impossible_bridge.zip](https://github.com/prusa3d/PrusaSlicer/files/15330413/impossible_bridge.zip)

**Describe the solution you'd like**
SuperSlicer have a solution for this. Setting is called "No perimeters on bridging areas". 
[image]
It works like this.
[image]
It would work even better if there would be option to limit perimeter count on bridge areas.

**Describe alternatives you've considered**
One can work around this problem by adding adding modifier with perimeters settings to bridge area, but it is troublesome. Modifier have to be 1 layer thin and have to be located exactly at bridge bottom layer.

thisloginalsotaken, 2024-05-16:
No, that is not it. In this example, project have "4 perimeters" setting, and supporing wall is only wide enough to place 3. Even in this image, 2 perimeters closest to bridge are hanging in the air.

tastyratz, 2024-06-08:
Doing this could potentially resolve https://github.com/prusa3d/PrusaSlicer/issues/9074 which would be HUGE since Arachne can't currently bridge over 1 perimeter.

shengyan, 2024-07-01:
> Doing this could potentially resolve #9074 which would be HUGE since Arachne can't currently bridge over 1 perimeter. I wonder if it would cut it? Would require testing. Edit: I just went to test this really cool looking feature BUT it looks like it's disabled for Arachne and only available for classic wall generation.

It works with Arachne, but the UI grey it out, so you cannot enable it under Arachne mode. You need to turn on classic mode first to enable it and then switch back to Arachne mode to use it.

tastyratz, 2024-07-08:
It does seem to work and like that issue was resolved (grayed out for Arachne) in the most recent built.
I believe this will significantly compensate for #9074 and largely band-aid that massive issue with Arachne.

tastyratz, 2026-09-18:
Merging this upstream has significant value to prusaslicer and downstream forks. Please do not close this feature request.