prusa3d/PrusaSlicer #12723: Collision of supports with objects

zburget, 2024-05-16:
When printing multiple objects that use organic supports and the objects are placed close together, the supports and adjacent objects collide. In the event of a collision, wouldn't it be possible to automatically "slide" the objects so that the collisions are resolved?

It is especially unpleasant when the objects on the mat are placed automatically and the slicer places them so close to each other that a collision occurs.

u89djt, 2024-05-16:
If you combine all the objects (select them all and right click) the supports will be generated for the ensemble and not crash into each other. You'll often see a given tree span more than one of your object.

zburget, 2024-05-17:
And this is an even better solution - to have an option something like "Generate supports for all objects together".
This would be especially useful when printing multiple instances of an object.

u89djt, 2024-05-17:
I'd be confident they're aware of this as a "nice to have". There's a software engineering burden; spawning new records of everything independent of the working set and deleting those when it's done without screwing everything up. Given that we can do this reliably ourselves it seems unlikely to outrank solving substantive errors and introducing new functionality that we can't do ourselves. It might not be complex per se, just asking for the usual kinds of trouble - have a look at the code base and see if you can spot a clean way of doing it.

closed 2026-09-29 by prusaslicer-bot[bot]