prusa3d/PrusaSlicer #12724: Prohibition of crossing perimeters

zburget, 2024-05-16:
Even in a situation where the perimeter crossing ban is set, the extruder needs to cross the perimeters - in a situation where printing is done in several places in one layer, it simply has to move to the next printing place after printing the filling. In this case, the extruder takes the shortest route and crosses the perimeters, perhaps even in a not very suitable place, and strings unnecessarily.
Wouldn't it be possible to add the ability to define where to run the extruder over the perimeters (eg at the back side) similar to how you can define (draw) the location of supports or the position of a seam? If such a place for the extruder to pass through the perimeters was defined in the given layer and object, the extruder would leave the object at this place. If it was not defined, it would exit anywhere (= the shortest path to the next object).

u89djt, 2024-05-17:
Just adding a note that mergeing everything into a single object doesn't solve this one.
[image]

Tupson444, 2026-09-19:
still relevant