prusa3d/PrusaSlicer #12729: No sparse layers causes crash if error during filament loading

andyylin, 2024-05-18:
### Description of the bug

I just ran a print using no sparse layers to avoid unnecessary waste. However, upon loading my third filament, the MMU had issues loading, and the extruder started to move toward the "standby" position for filament issues, causing a crash into the print.

My guess is that the print was at the z-height of the no sparse layer wipe tower, and when the filament issue triggered a move, the extruder didn't first move to the height of the print before moving to the standby postion.

(Forgot to take pictures...I hope the description is complete.)

### Project file & How to reproduce

[Award v1.1 - 5S.zip](https://github.com/prusa3d/PrusaSlicer/files/15359033/Award.v1.1.-.5S.zip)
[image]

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+win64

### Operating system

Windows 11

### Printer model

Prusa MK3S+ with MMU2S

SamuelFarid, 2025-04-16:
I also have faced the same problem and I am trying to find a solution. Have you ever found a solution to it? I thought of adding a custom GCODE to increase the Z axis value with tool change so if it failed and decided to home it will be far enough. I still need to try it out.

SachCZ, 2025-04-16:
Hello, this is the current limitation of the no sparse layers option and a big reason why it is still labeled as EXPERIMENTAL. It is the limitation of the current implementation, which I think is still better than nothing.

In the future there hopefully might be some improvements.

closed 2026-09-29 by prusaslicer-bot[bot]