prusa3d/PrusaSlicer #12732: Firmware Type question during PrusaSlicer installation

fscheps, 2024-05-19:
### Description of the bug

While installing version 2.7.4 on the Mac with an M1 Ultra chip, I was asked to select which printer I was going to use, I selected the Prusa XL 2T I have. A couple of steps later I was asked which type of firmware my printer uses. I have no clue, but the default selection is "RepRap/Sprinter". This should be pre-filled with the default based on the selected printer on previous steps, or at least a reference to what firmware is the one used by the Prusa printers without pre-selecting anything and avoiding potential future issues.

### Project file & How to reproduce

Please see comment above.

[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

MacOS 14.4.1

### Printer model

Prusa XL 2 Toolheads

rtyr, 2024-05-20:
You don't have to fill-in custom printer section. It's meant for custom non-Prusa printers.

cosmix, 2024-06-01:
@rtyr you’re right, however the user experience could definitely be improved, and it is, apparently, confusing to some users. I would file this as improvement, low priority.

closed 2026-09-29 by prusaslicer-bot[bot]