prusa3d/PrusaSlicer #12735: 7.2.4 does not compile with boost-1.85.0: fatal error: boost/filesystem/convenience.hpp: No such file or directory

fordfrog, 2024-05-20:
### Description of the bug

```
FAILED: src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/RemovableDriveManager.cpp.o 
/usr/bin/x86_64-pc-linux-gnu-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_CHRONO_DYN_LINK -DBOOST_CHRONO_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_LOCALE_DYN_LINK -DBOOST_LOCALE_NO_LIB -DBOOST_LOG_DYN_LINK -DBOOST_LOG_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DLIBNEST2D_GEOMETRIES_libslic3r -DLIBNEST2D_OPTIMIZER_nlopt -DLIBNEST2D_STATIC -DLIBNEST2D_THREADING_tbb -DOPENVDB_ABI_VERSION_NUMBER=11 -DOPENVDB_USE_DELAYED_LOADING -DSLIC3R_CURRENTLY_COMPILING_GUI_MODULE -DSLIC3R_GUI -DTBB_USE_CAPTURED_EXCEPTION=0 -DUNICODE -DUSE_TBB -DWXINTL_NO_GETTEXT_MACRO -DWXUSINGDLL -D_FILE_OFFSET_BITS=64 -D_UNICODE -D__WXGTK__ -DwxNO_UNSAFE_WXSTRING_CONV -DwxUSE_UNICODE -I/usr/include/dbus-1.0 -I/usr/lib64/dbus-1.0/include -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999_build/src/platform -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/hidapi/include -I/usr/include/gtk-3.0 -I/usr/include/pango-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/webp -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/atk-1.0 -I/usr/include/elogind -I/usr/include/fribidi -I/usr/include/libpng16 -I/usr/include/pixman-1 -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/gio-unix-2.0 -I/usr/lib64/libffi/include -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/libmount -I/usr/include/blkid -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999_build/src/libslic3r -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/libnest2d/include -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/miniz -I/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/glu-libtess/include -isystem /usr/include/eigen3 -isystem /usr/lib64/wx/include/gtk3-unicode-3.2-gtk3 -isystem /usr/include/wx-3.2-gtk3 -isystem /var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/libigl -isystem /usr/include/LibBGCode -isystem /usr/include/Imath -isystem /usr/include/nanosvg  -DNDEBUG -march=alderlake -mabm -mno-cldemote -mno-kl -mno-pconfig -mno-sgx -mno-widekl -mshstk --param=l1-cache-line-size=64 --param=l1-cache-size=48 --param=l2-cache-size=36864 -O2 -pipe -frecord-gcc-switches -fext-numeric-literals -Wall -Wno-reorder -pthread -std=gnu++17 -fsigned-char -Werror=return-type -Wno-ignored-attributes -Wno-unknown-pragmas -MD -MT src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/RemovableDriveManager.cpp.o -MF src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/RemovableDriveManager.cpp.o.d -o src/slic3r/CMakeFiles/libslic3r_gui.dir/GUI/RemovableDriveManager.cpp.o -c /var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/slic3r/GUI/RemovableDriveManager.cpp
/var/tmp/portage/media-gfx/prusaslicer-9999/work/prusaslicer-9999/src/slic3r/GUI/RemovableDriveManager.cpp:38:10: fatal error: boost/filesystem/convenience.hpp: No such file or directory
   38 | #include <boost/filesystem/convenience.hpp>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Project file & How to reproduce

install boost-1.85.0 and try to compile prusaslicer

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

7.2.4 and tried also the latest from the repo

### Operating system

gentoo linux

### Printer model

MK4

lukasmatena, 2024-05-20:
Thanks. This will be fixed in PrusaSlicer 2.8.0: https://github.com/prusa3d/PrusaSlicer/pull/12652.
Closing.

closed 2024-05-20 by lukasmatena

lukasmatena, 2024-05-21:
@fordfrog It may take about a month, but do not quote me on that anywhere.

fordfrog, 2024-05-21:
> @fordfrog It may take about a month, but do not quote me on that anywhere.

thanks. i guess we will have to patch 2.7.4 in gentoo to resolve this issue before 2.8.0 is released.