prusa3d/PrusaSlicer #12736: Brim on corners

tk-notify, 2024-05-20:
Currently I always have to add manually STL circles to get brims at the corners, as known from SuperSlicer.
With large prints (close to 500mm in length) I have problems with brims on the full length regarding bed adhesion. So I need to limit the brim to the edges.

It would be great to have the same option as in SuperSlicer. There it is possible to limit the brim on outer corners.
The reason why I am not using SuperSlicer anymore is due to the much better presets and specific support for V-Core (Ratrig) in Prusa. And in SuperSlicer I am missing the brilliant infill pattern "Cubic".

As in SuperSlicer I would like to have circles at outer corners, starting with smallest radius from the model corner radius and ending with max. radius = model corner radius + brim width

Thanks,
Tobias

neophyl, 2024-05-20:
Before opening new issues please please search for existing ones.  For example
https://github.com/prusa3d/PrusaSlicer/issues/2954
https://github.com/prusa3d/PrusaSlicer/issues/4568 where the superslicer implementation is referenced.   There are others too.

tk-notify, 2024-05-20:
Closed, with respect to already open requests. Most relevant for me: https://github.com/prusa3d/PrusaSlicer/issues/2954

closed 2024-05-20 by tk-notify

Snuff1eupagus, 2024-08-31:
@tk-notify 
Maybe this will help you to achieve your goals with Brims

Selective Blocking
https://www.youtube.com/watch?v=6CRqNk_JUj4

Selective Adding
https://www.youtube.com/watch?v=dhsbMmosbLU&t=1s

Regards