prusa3d/PrusaSlicer #12738: Transition nozzle heat up X-layers

FlailAway, 2024-05-21:
With fast printers now being a "thing" I have been on a quest to utilize the fastest speed with a good finish. To this end I have been increasing Temperature rather than slowing down for under-extrusion issues.

This is working well overall, but not for the first few layers as there is  lot of stringing with the over-hot "runny" filament. At a defined 200mm/s and 260C the bulk of a PETG print is great.

I tried setting First Layer to 235C, but along with fast printers come fast heating nozzles. Within 15-seconds or so the nozzle is up to the 260C temp, way too "runny" and still only on the second layer

So to the question, instead of just a First-Layer Temp. can we make that a selected number of layers to transition to the secondary Temp?  

e.g. For "5" layers: 260C - 235C = 25C. Then 25 / 5 (layers)  = 5C to bump the Temp by that figure for each layer.

OR, is there some other **existing** way I can do this?

u89djt, 2024-05-22:
Here you go: https://help.prusa3d.com/article/insert-pause-or-custom-g-code-at-layer_120490

neophyl, 2024-05-22:
You can also add some conditionals into the layer change section of the Printer profile.
For example this thread does something similar but for changing flow https://forum.prusa3d.com/forum/prusaslicer/how-to-do-arithmetic-inside-a-condition/  you can just as easily change temps and add more conditions.  

M104 S{temperature[0]-50} ; for example to set it to the filament temperature -50 (or whatever you want) and then use other -xx values to get closer and closer to your final temp at the layer you want.

FlailAway, 2024-05-25:
Thanks, but that seems to be per print/slice and not thinking "big-picture."

I now only have fast printers and all prints would need the gcode modified every time. That would get old fast tweaking gcode for every development iteration when designing items.

Far better to have the option in the slicer as I would leave it set at x-layers and let it calculate the rise based on the start (currently First-Layer Temp) to the desired running temp.

neophyl, 2024-05-25:
You add it to your printer profile and then slice with that. It then automatically gets added when slicing with that printer. 
Of course if you want to wait for a new feature to be added when you could be doing it now then that’s up to you.

FlailAway, 2024-05-31:
Thanks, I understood that, but the temp Start/Spread/Main is different for each filament. I'd need 10- different printer profiles. The flow is more of a global thing.

Perhaps if there was "before-layer-change-GCode" or "after-layer-change-GCode" in the Filament section it would work for now.

FlailAway, 2024-06-06:
Just noticed Cura already has this. Not enough to make me switch, just sayin'. :)

closed 2026-09-29 by prusaslicer-bot[bot]