prusa3d/PrusaSlicer #12741: Sequential Printing with multiple extruders doesnt properly set is_extruder_used

jfetzener, 2024-05-22:
### Description of the bug

When sequential printing multiple objects with extruder changes, only the tools in the first object are treated as used, meaning the tools only used in the first object will have is_extruder_used[] as true. This causes the other extruders not to be primed, have their temperature set, nozzle diameter output in gcode, or any other custom gcode output that relies on is_extruder_used[].

### Project file & How to reproduce

[seq_print_bug.zip](https://github.com/prusa3d/PrusaSlicer/files/15404918/seq_print_bug.zip)

Sequential printing enabled, notice only the extruders used in the first object are primed. Disabling sequential printing causes all used extruders to be primed.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 10 Enterprise

### Printer model

XL5IS

jfetzener, 2024-05-22:
[image]

[image]

u89djt, 2024-05-23:
@jfetzener hoping to benefit from your insights into gcode/tool management. Have you noticed any unproductive pauses in general multi tool printing? In many jobs I'll see the tool and changer idle for a few seconds, and the display info shows waiting for temperature. I'm going to try looking through plain text gcode from some of the jobs where this happens, but if you happen to be aware of it, could I ask you to highlight anything you know that might be relevant please?

closed 2026-09-29 by prusaslicer-bot[bot]