prusa3d/PrusaSlicer #12743: Headbed does not turn off after Print is finished

ikildkenny, 2024-05-23:
### Description of the bug

Printer: Kingroon KP3S
Issue was not present in gcodes generated with Slicer Version 2.7.0 and the same profile

After the print is finished, the heatbed holds its temperature and has to be shut off manually.

### Project file & How to reproduce

M140 S0 ; turn off heatbed

is present in gcode - still not turning off for files sliced with 2.7.4

no project file at hand

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

windows 10

### Printer model

Kingroon KP3S

u89djt, 2024-05-23:
https://www.namu3d.com/2021/03/11/kingroon-kp3s-start-g-code/
Looks like M140 is the extruder temp and the bed temp is M190?
(random fellow user watching issues go past)

rtyr, 2024-05-23:
> Looks like M140 is the hot end temp and the bed temp is M190?
both of those are used for bed temp (one without waiting for target temp, other with waiting)

> M140 S0 ; turn off heatbed
> is present in gcode - still not turning off for files sliced with 2.7.4

If this g-code command is really present in the g-code and printer ignores it (or does not support it), than the issue is not related to the slicer.

Anyway, all we can to here is check the 3MF project file, if provided.

ikildkenny, 2024-05-23:
3mf with issue added

rtyr, 2024-05-23:
I checked and the M140 S0 command is present in the output g-code. You may try to remove G4 command before it, maybe your FW have some issues with it.

I don't see any issue related to PrusaSlicer.

ikildkenny, 2024-05-24:
The thing is - the issue did not and does not occur with files sliced with 2.7.0 
[stringtest.zip](https://github.com/prusa3d/PrusaSlicer/files/15427540/stringtest.zip)
e.g. this one

ikildkenny, 2024-05-24:
as further reference two gcodes, one sliced with an earlier version (works fine), one sliced with 2.7.4 (suffix 274)
[gcodes.zip](https://github.com/prusa3d/PrusaSlicer/files/15430091/gcodes.zip)

neophyl, 2024-05-24:
The gcode to command the bed to turn off is in both.  So the printer should be turning it off.  I agree with rtyr that this is a printer firmware issue technically.
The major difference I can see at the end of the gcode file is the new 274 version also includes a very long single line comment for the objects_info
`; objects_info = {"objects":[{"name":"Warmachine-Hordes_40mm_full.stl id:1 copy 0","polygon":[[66.943,88.014],[68.897,88.208],[70.821,88.590],[72.699,89.160],[74.513,89.912],[76.243,90.836],[77.875,91.928],[79.393,93.172],[80.781,94.560],[82.027,96.076],[83.117,97.710],[84.041,99.440],[84.793,101.254],[85.363,103.132],[85.745,105.056],[85.939,107.010],[85.939,108.972],[85.745,110.926],[85.363,112.850],[84.793,114.728],[84.041,116.542],[83.117,118.272],[82.027,119.904],[80.781,121.422],[79.393,122.810],[77.875,124.056],[76.243,125.146],[74.513,126.070],[72.699,126.822],[70.821,127.392],[68.897,127.774],[66.943,127.968],[64.981,127.968],[63.027,127.774],[61.103,127.392],[59.225,126.822],[57.411,126.070],[55.681,125.146],[54.047,124.056],[52.531,122.810],[51.143,121.422],[49.899,119.904],[48.807,118.272],[47.883,116.542],[47.131,114.728],[46.561,112.850],[46.179,110.926],[45.985,108.972],[45.985,107.010],[46.179,105.056],[46.561,103.132],[47.131,101.254],[47.883,99.440],[48.807,97.710],[49.899,96.076],[51.143,94.560],[52.531,93.172],[54.047,91.928],[55.681,90.836],[57.411,89.912],[59.225,89.160],[61.103,88.590],[63.027,88.208],[64.981,88.014]]},{"name":"Warmachine-Hordes_50mm_full.stl id:2 copy 0","polygon":[[110.264,55.743],[112.706,55.983],[115.112,56.461],[117.460,57.173],[119.726,58.113],[121.890,59.269],[123.930,60.633],[125.826,62.187],[127.560,63.923],[129.118,65.819],[130.480,67.859],[131.636,70.023],[132.576,72.291],[133.288,74.637],[133.766,77.043],[134.006,79.485],[134.006,81.939],[133.766,84.381],[133.288,86.787],[132.576,89.135],[131.636,91.401],[130.480,93.565],[129.118,95.605],[127.560,97.501],[125.826,99.235],[123.930,100.793],[121.890,102.155],[119.726,103.311],[117.460,104.251],[115.112,104.963],[112.706,105.441],[110.264,105.681],[107.810,105.681],[105.368,105.441],[102.962,104.963],[100.616,104.251],[98.348,103.311],[96.184,102.155],[94.144,100.793],[92.248,99.235],[90.512,97.501],[88.958,95.605],[87.594,93.565],[86.438,91.401],[85.498,89.135],[84.786,86.787],[84.308,84.381],[84.068,81.939],[84.068,79.485],[84.308,77.043],[84.786,74.637],[85.498,72.291],[86.438,70.023],[87.594,67.859],[88.958,65.819],[90.512,63.923],[92.248,62.187],[94.144,60.633],[96.184,59.269],[98.348,58.113],[100.616,57.173],[102.962,56.461],[105.368,55.983],[107.810,55.743]]},{"name":"Warmachine-Hordes_30mm_full.stl id:0 copy 0","polygon":[[65.870,52.026],[67.334,52.172],[68.778,52.458],[70.186,52.886],[71.546,53.450],[72.844,54.144],[74.068,54.960],[75.206,55.894],[76.248,56.936],[77.182,58.074],[77.998,59.298],[78.694,60.596],[79.256,61.956],[79.684,63.364],[79.970,64.808],[80.116,66.272],[80.116,67.746],[79.970,69.210],[79.684,70.654],[79.256,72.062],[78.694,73.422],[77.998,74.720],[77.182,75.944],[76.248,77.082],[75.206,78.124],[74.068,79.058],[72.844,79.874],[71.546,80.570],[70.186,81.132],[68.778,81.560],[67.334,81.846],[65.870,81.992],[64.396,81.992],[62.932,81.846],[61.488,81.560],[60.080,81.132],[58.720,80.570],[57.422,79.874],[56.198,79.058],[55.060,78.124],[54.018,77.082],[53.084,75.944],[52.268,74.720],[51.574,73.422],[51.010,72.062],[50.582,70.654],[50.296,69.210],[50.150,67.746],[50.150,66.272],[50.296,64.808],[50.582,63.364],[51.010,61.956],[51.574,60.596],[52.268,59.298],[53.084,58.074],[54.018,56.936],[55.060,55.894],[56.198,54.960],[57.422,54.144],[58.720,53.450],[60.080,52.886],[61.488,52.458],[62.932,52.172],[64.396,52.026]]}]}`

Its possible the printer cant handle that due to its length.  Personally I would edit the 274 code to remove that and see if it will then cool down.  If it does then you have narrowed it down.

ikildkenny, 2024-05-24:
> The gcode to command the bed to turn off is in both. So the printer should be turning it off. I agree with rtyr that this is a printer firmware issue technically. The major difference I can see at [...]

sadly no joy - same behaviour (tbh - unsurprisingly, but still a decent idea)

I noticed that the estimated time in 2.7.4 is 2 mins shorter for the same files - maybe something in the optimized gcode messes with stuff here?

closed 2026-09-29 by prusaslicer-bot[bot]