prusa3d/PrusaSlicer #12748: Malformed slicing

drbobbob, 2024-05-24:
### Description of the bug

I designed a part in Fusion and imported it into prusa slicer.  The model preview appears correct but when I looked at the sliced file the first layer was missing a large chunk.

If I set elephants foot compensation to 0 it appears that the file slices correctly.

### Project file & How to reproduce

[PrusaSlicerSTLIssue.zip](https://github.com/prusa3d/PrusaSlicer/files/15440256/PrusaSlicerSTLIssue.zip)
Zip contains the 3mf file saved from prusaslicer which contains the bad model.
FusionPart.png is a screenshot of the part in Fusion.
Layer1.png is what the first layer gcode looks like.
Layer2.png is what the second layer looks like, note the large triangular area that should have been present in layer 1. 

To reproduce, load the 3mf file and slice.  An STL or 3mf file as exported from Fusion can also be provided.

I suspect the fact that there are two small vertical lines on the face of the part caused the trangulation to have some edges really close together which confused the slicer potentially when it was attempting to do 

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 11 home 10.0.22631

### Printer model

Prusa Mk4

u89djt, 2024-05-25:
Fellow Fusion 360 user here - if you want a second pair of eyes on the original model, go ahead and post it. From initial experiementation with importing and converting the mesh prismatically and facetted, with or without repairing first, there does seem to be something whacky about the region around the tiny triangle in the base. 
Facetted conversion without repair gives a body where all the angles are nice whole numbers. If you repair or convert prismatically, it either fails, or gives small angle errors, including in the base between faces. That's presumably something the slicer has to try to see through to something it can print in the first layer.

drbobbob, 2024-05-26:
Here's my fusion file that produced the error in slicing.  
[Saw Angle Guide Fusion File.zip](https://github.com/prusa3d/PrusaSlicer/files/15447624/Saw.Angle.Guide.Fusion.File.zip)

It wasn't critical that those channels go all the way to the end.  This is just a piece to give a visual reference when sharpening a chainsaw chain.  I was able to work around the issue just fine.  I'm also terrible about getting the right planes selected to start my part in fusion.  The intention of the design was to be printed with the grooves on bottom so that magnets can be glued into the holes.  Its also intended to have a color change so that there is nice contrast and you can see into the grooves.

u89djt, 2024-05-26:
Cool. As a clean dismount, here's a fix:
[image]
[image]
Fusion and 3mf files
[excerpt and hack 3mf and fusion.zip](https://github.com/prusa3d/PrusaSlicer/files/15447778/excerpt.and.hack.3mf.and.fusion.zip)
It seems like ideally the slicer would cope with a mesh that appears to be a coherent solid, but I don't know how the representation can in theory or does in practice here deal with sharp meeting edges. Obviously those can't actually be printed by anything - I guess the model should ideally state what is intended in the print. That's what made me try the square pipe to cut it. That the edges in the main face are dealt with is a different matter that I have no ideas about (yet).
Someone probably knows the answer...

drbobbob, 2024-05-26:
For those curious, here is the final model that I was attempting to make.  [Chainsaw Angle Guide](https://www.printables.com/model/892367-chainsaw-sharpening-guide-different-magnets)

closed 2026-09-29 by prusaslicer-bot[bot]