prusa3d/PrusaSlicer #12750: Linux -- Application Font size ignored in Preview Legend, Tip of the Day, and other viewport areas -- unreadably small on a 4k laptop

x0pherl, 2024-05-25:
### Description of the bug

Changing the Application Font size doesn't impact the size of text within the main viewport -- application of the day, legend, warnings, "slicing finished" etc are all unreadable on a 4k laptop

### Project file & How to reproduce

start up the application on a 4k screen laptop and look at the message of the day
-OR-
slice any object and view the legend
[image]
[image]

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4

### Operating system

OpenSUSE Tumbleweed in KDE Desktop

### Printer model

mk4

m8ram, 2025-06-01:
Same here, PrusaSlicer 2.9.2 on a 27" monitor at 3840x2160 resolution.
Tooltips and warnings/notifications and the Layers dialog are impossible to read.

[image]

[image]

[image]

I have increased the font size of the GUI but this has no effect on these items.

StrNamePete, 2025-09-07:
I have the exact same issue also running Fedora 42 Workstation (Gnome) with Wayland. My Prusa Slicer is version 2.9.2 (flatpak as well).

closed 2026-09-29 by prusaslicer-bot[bot]