prusa3d/PrusaSlicer #12752: Skirt material extruder selection does not appear to work correctly

bacomatic, 2024-05-26:
### Description of the bug

I am printing an object in ASA and using PLA as a support interface material.

Print Settings:
- Multiple Extruders
  - Support material/raft/skirt extruder: 1 (extruder loaded with ASA)
  - Support material/raft interface: 2 (extruder loaded with PLA)

When I set skirt loops to 1 and set Draft shield to "Enabled", it generates a draft shield around the model but instead of being solid ASA, it has layers of PLA mixed in. Needless to say this will not work as it is guaranteed to separate during printing (I did test this...)

I expected "Support material/raft/skirt extruder" to force the skirt to only be printed with the selected extruder, as it does for the support material itself.

### Project file & How to reproduce

[R50-R53_phonemount_LHD_ballmount-bossed-draft.zip](https://github.com/prusa3d/PrusaSlicer/files/15445977/R50-R53_phonemount_LHD_ballmount-bossed-draft.zip)

To reproduce just load the project and click slice.

A skirt should be generated with bands of the incorrect material on the same layers as the support interface material.

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 10

### Printer model

Prusa XL 5 toolhead

metaxis, 2024-06-23:
I'm seeing the same with a multiple extruder setup and a simple skirt. The more loops I add, the more materials the slicer picks, up to 3. It is ignoring the Support material/raft extruder setting.

u89djt, 2024-06-23:
Fellow user agreeing that we need coherent skits with a brim. In the meantime, there might be folk reading this who would benefit from the simplest workaround available of setting up your own skirt. Obviously there are many situations in which this would be a pain, but you can also make a detailed skirt in whatever the object is designed in. Folk could even include a bespoke skirt in an object post to printables if it's worth the effort.
[tailored_R50-R53_phonemount_LHD_ballmount-bossed-draft.zip](https://github.com/user-attachments/files/15944145/tailored_R50-R53_phonemount_LHD_ballmount-bossed-draft.zip)
[image]

JeremiahGillis, 2025-07-06:
I encountered this issue today with PrusaSlicer 2.9.2. It would be nice to see this fixed.

JeremiahGillis, 2026-09-22:
Any one here still know if this is a problem? I have not tested this again recently.

bacomatic, 2026-09-22:
My printer is buried in my shed so I haven't been able to test. Though it also shows up in the slicer, you can see the different materials in the shroud.