prusa3d/PrusaSlicer #12755: A G-code path goes beyond plate boundaries. (Treesupport)

tom-321, 2024-05-27:
### Description of the bug

If I use treesupport this happens:
[image]

### Project file & How to reproduce

Tree support ignores the boundaries of the bed.

### Checklist of files included above

- [ ] Project file
- [X] Screenshot

### Version of PrusaSlicer

Version 2.7.4+MacOS-x64

### Operating system

macOS Sonoma 14.4.1

### Printer model

P1P

tom-321, 2024-05-27:
Ok solved it, just add a little cube in front: 
[image]

kubispe1, 2024-05-28:
Hi, Just a quick update to let you know that this issue is unfortunately still unresolved. It appears to be a duplicate of issue #849

closed 2024-05-28 by kubispe1