prusa3d/PrusaSlicer #12756: Instant crash when type-changing support blocker primitive to modifier

fionnb, 2024-05-27:
### Description of the bug

PrusaSlicer supports modifying the model with primitives of various types. There also is a menu option to change this type. But if you try to actually use that, the application crashes.

### Project file & How to reproduce

[3D4U_Coffee_Clip_Large_V3.1_HC_XL_CF-PLA.zip](https://github.com/prusa3d/PrusaSlicer/files/15456883/3D4U_Coffee_Clip_Large_V3.1_HC_XL_CF-PLA.zip)

1. Load the project
2. right-click on "Supportblocker 3" in the panel right of the model display for a context menu
3. click "change type"
4. click "modifier"
5. click "OK"
6. instant crash

### Checklist of files included above

- [X] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4

### Operating system

Arch Linux

### Printer model

Prusa XL 5tool

kubispe1, 2024-05-28:
Hi, thanks for report. But, I cannot reproduce on Win 10 with PS 274.

Jan-Soustruznik, 2024-05-28:
Hello, @fionnb, thank you for your report, but I cant also reproduce you reported issue on my Linux(Ubuntu 20.04.6 LTS). Could you provide some more info about you OS/HW?, GTK2 or GTK3?

Thank you, have a nice day

fionnb, 2024-05-28:
Arch Linux on Ryzen 7840U, kernel 6.8.9, KDE6 / Plasma6, GTK3

This is the output when I reproduce the bug starting from the shell:

```
$ prusa-slicer

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.653: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.653: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.655: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.656: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.658: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.658: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.886: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.887: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.888: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.888: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.931: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.931: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.940: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.940: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.941: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.941: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.941: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.941: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.941: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.942: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.942: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.942: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:55:57.943: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
[... 16 lines cut ...]

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_widget_set_size_request: assertion 'width >= -1' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

[... cut ...]
(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed

(prusa-slicer:3096221): Gtk-CRITICAL **: 18:56:11.668: gtk_cell_layout_get_cells: assertion 'GTK_IS_CELL_LAYOUT (cell_layout)' failed
/usr/include/c++/13.2.1/bits/stl_vector.h:1125: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slic3r::ObjectID; _Alloc = std::allocator<Slic3r::ObjectID>; reference = Slic3r::ObjectID&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
Abort
```

closed 2026-09-29 by prusaslicer-bot[bot]