prusa3d/PrusaSlicer #12759: Very slow or frozen G-code generation

wizzlebippi, 2024-05-27:
### Description of the bug

I just finished installing the MMU3 on my MK4 and have discovered that there are some files that simply do not finish G-code generation.  The slicing works up until the "Generating G-code." step at 90% and never progresses.  I let one run for about 20 hours to check to see if it was just slow, but no luck.  I've attached one problematic project file as an example. 

### Project file & How to reproduce

[Baby Turtle.zip](https://github.com/prusa3d/PrusaSlicer/files/15458885/Baby.Turtle.zip)
Open and slice this project file. 

### Checklist of files included above

- [ ] Project file
- [ ] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 11

### Printer model

Prusa MK4 with MMU3

u89djt, 2024-05-27:
Weird. If you rotate it, it finishes slicing. I only tried this small angle and 90 degrees.
[image]

wizzlebippi, 2024-05-27:
That makes absolutely no sense, but I can confirm that it works.

u89djt, 2024-05-27:
I edited the comment ^^ 90 degrees counterclockwise /doesn't/ terminate. It doesn't make obvious sense to us outside the code, but this begins to suggest that there's a graunch in some trigonometry somewhere or one of a million other things that happen in coding challenges.

u89djt, 2024-05-27:
If you change seam position to "random", it succeeds in the orientation in the 3mf file.

wizzlebippi, 2024-05-28:
[image]
[color rocket (1).zip](https://github.com/prusa3d/PrusaSlicer/files/15463039/color.rocket.1.zip)

Found another one.  Tried messing around with seams for a while, including deleting the painted on seams, as well as rotating slightly.  No joy.

u89djt, 2024-05-28:
Aha, so for this one, and for the turtle in your original file, deactivating "Staggered inner seams" without changing anything else allows it to complete. See if that does you any good in other files?

wizzlebippi, 2024-05-28:
Confirmed that disabling "Staggered Inner Seams" allows g-code generation to complete.  I'll take a look at my other files and see if I can find one that isn't fixed by disabling that.  Thanks

closed 2026-09-29 by prusaslicer-bot[bot]