prusa3d/PrusaSlicer #12764: 2.7.4 Generated Gcode without z-lift causing nozzle crashes

TheApsik, 2024-05-31:
### Description of the bug

I have set Travel lift > Lift height to 10mm and it doesn't work when it move from one place to another causing to hit printed parts.
[image]
[image]

### Project file & How to reproduce

[wheel.zip](https://github.com/user-attachments/files/15510227/wheel.zip)

### Checklist of files included above

- [X] Project file
- [X] Screenshot

### Version of PrusaSlicer

2.7.4+win64

### Operating system

Windows 10

### Printer model

Voron 2.4r2

rtyr, 2024-05-31:
You have "Only retract when crossing perimeters" parameter enabled, therefore retractions/z-lifts are not performed in such cases.

TheApsik, 2024-05-31:
> Only retract when crossing perimeters

It is confusing.. Two parameters located in different places are related with each other.. Shouldn't this relation be explicit. I would never thought it is related..

TheApsik, 2024-07-11:
Will there be done any changes to make this behaviour explicit?

Overruler, 2024-09-13:
Setting "Retraction length" to 0 causes "Lift height" and "Maximum ramping lift" values to be ignored, as if they had been set to 0. "Minimum travel after retraction" does the same for the affected travel moves. OTOH, setting "Retraction length" to 0.00001 causes "Lift height" and "Maximum ramping lift" values to be applied correctly.

Tested in PrusaSlicer-2.8.1-rc1+win64-202409121245.

ClayShoaf, 2025-09-20:
Same thing is happening for me on version 2.9.3

@Overruler 
> Setting "Retraction length" to 0 causes "Lift height" and "Maximum ramping lift" values to be ignored, as if they had been set to 0. "Minimum travel after retraction" does the same for the affected [...]

I do not have any of those set to 0.

[image]

[image]

ClayShoaf, 2025-09-20:
Solved!

I went to filament settings and unchecked "use ramping lift" there. Not sure why this setting exists for the printer and the filament, or why changing one would not automatically change the other, but it looks like it's going to work now.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]